OpenShock/Firmware #19: Built firmwares are broken

LucHeart, 2023-10-04:
Firmeware bins compiled by github actions dont seem to work at all

hhvrc, 2023-10-04:
Firmware*

redmushie, 2023-10-04:
I think it's time to add issue templates because this issue contains zero useful information for me to even understand what you tried, let alone reproduce it ._.

LucHeart, 2023-10-04:
Try to flash a built artifact, it doesnt work, straightup no output, nothing-

LucHeart, 2023-10-04:
e.g. https://github.com/OpenShock/Firmware/suites/16850240278/artifacts/962381648

redmushie, 2023-10-04:
> a build artifact

What artifact?

> try to flash

What are you flashing? How are you flashing? To what board are you flashing?

LucHeart, 2023-10-04:
The built artifacts that can be found under the actions section on github obviously. Latest develop OpenShock_Wemos-D1-Mini-ESP32. Prettysure all are broken tho.

Am trying to flash a D1 mini, esptool, esptooljs, doesnt matter.

redmushie, 2023-10-04:
Yes, it matters.

Artifacts are fine for me.
Output is fine for me.

If you don't specify exactly what commands you're using with their exact arguments, there's no way for me to even _try_ to reproduce what you're seeing, and I'd close the issue as non-reproducible.

LucHeart, 2023-10-04:
There is not more information to be give ngl. here have a list to reproduce, its as I already described.

1. Download latest D1 mini firmware from artifacts section (develop)
2. Connect your ESP-32 (D1 mini) via usb
3. Go to https://espressif.github.io/esptool-js/
4. Connect to your board
5. Select firmware bin and set offset to 0x0
6. Flash
7. Not alive after reset

redmushie, 2023-10-04:
Thanks. Step 3/4/5/6 is what I needed.

I went to check the latest artifacts on the `develop` pipeline but they've already expired:
[image]

(Expiry is set to 1 day by default. I'll probably change this soon)

Can you tell me where you got the exact artifact from? I.e. not the download link but a link to the place _where you found_ the download link. Thanks

LucHeart, 2023-10-04:
Dafuq? At what run are you looking at? Am looking at latest develop. Link here: https://github.com/OpenShock/Firmware/actions/runs/6400794452

LucHeart, 2023-10-04:
nvm my bad, its the captive portal fix branch - had the same on develop tho

redmushie, 2023-10-04:
XD

redmushie, 2023-10-04:
I bumped retention to 7 days for the merged firmware artifacts on `develop`. Try those when it finishes building @LucHeart

LucHeart, 2023-10-04:
develop works now, maybe something on the cpative protal branch is broken. I will test more.

LucHeart, 2023-10-04:
Works aswell now - well sorta - tried erasing flashing before, that seemed to have done the trick.

Another issue that might need to be looked into, the LED doesnt flash, and just in general the captive portal is having issues with the frontend. - Forwarding to @hhvrc . Closing this issue for now.

closed 2023-10-04 by LucHeart