OpenShock/Firmware #38: Make a checklist for testing

hhvrc, 2023-10-09:
We need a checklist for testing release candidates.

@redmushie

redmushie, 2023-10-16:
# Release Candidate Per-Board Checklist

# Basics
- [ ] Image can be flashed to board
- [ ] Board outputs logging statements over serial (for debug builds)
- [ ] Startup header / `version` command prints expected version (`[major].[minor].[patch]-rc.[number]`)

## Serial
- [ ] Board outputs _any_ serial output
- [ ] Board accepts _any_ serial command (`help` or `version` suffices)
- [ ] `help` prints all commands
- [ ] `version` prints correct version
- [ ] `reset` restarts board
- ...

## Status LED

Choose whichever applies:

### GPIO Status LED
- [ ] LED lights up at all
- [ ] ...

### WS2812B Status LED
Currently unsupported.

# Captive portal

## Availability
- [ ] WiFi network appears (named `OpenShock-XX:XX:XX:XX:XX:XX`)
- [ ] WiFi network can be connected to
- [ ] Visiting `http://10.10.10.10/` while connected returns _any_ web page (blank or partial = OK!)

## Functionality
- [ ] Network list is populated automatically after page load
- [ ] Sync button (re)populates WiFi network list

### Connecting
- [ ] Clicking network shows "connect" modal
- [ ] Connect modal can be closed
- [ ] Connect modal can be submitted
- [ ] Connection attempt is made when modal is submitted
- [ ] Connection succeeds when correct credentials are used

# Connectivity
- [ ] Connects to `shocklink.net` by default (???)
- [ ] Can use pairing code to connect to server
- [ ] Shows up as online on server board list (depends on API, LiveControlGateway and WebUI)

# Shockers

## Settings
- [ ] `rmtpin` prints current RMT pin
- [ ] `rmtpin <pin>` changes current RMT pin (verify using `rmtpin`)

## Transmitting
- [ ] Transmitting any command works (vibrate or beep suffices)

redmushie, 2023-10-16:
Something like that @hhvrc ! Feedback welcome

redmushie, 2023-10-16:
Is that all? Surely I've forgotten some things that are testable @hhvrc . Or some things might be a bit too verbose/unnecessary?

redmushie, 2023-10-18:
Marking as done per #51

closed 2023-10-18 by redmushie