OpenShock/Firmware #40: [bug] ESP32-S3 Panics when connecting to WiFi network

hhvrc, 2023-10-09:
Happens when connecting to a WiFi network trough the Cative Portal

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420c3779  PS      : 0x00060b30  A0      : 0x820bc734  A1      : 0x3fceb0f0  
A2      : 0x3fceb2f0  A3      : 0x00000001  A4      : 0x3fceb2f0  A5      : 0x3fceb0f2  
A6      : 0x000000ca  A7      : 0x3fceb0f0  A8      : 0x00000000  A9      : 0x00000008  
A10     : 0x000000a5  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000010  
A14     : 0x00000068  A15     : 0x3fceb480  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x000000a5  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffb  

Backtrace: 0x420c3776:0x3fceb0f0 0x420bc731:0x3fceb480 0x42007d51:0x3fceb540 0x42008edd:0x3fceb9c0 0x420d0e89:0x3fcebb20 0x42030025:0x3fcebb50 0x42030051:0x3fcebb80 0x420d1ab9:0x3fcebba0 0x420306ab:0x3fcebbc0 0x420d1b8f:0x3fcebbf0 0x420309d9:0x3fcebc20 0x42030aa6:0x3fcebc50 0x42030c2e:0x3fcebc70 0x42030d62:0x3fcebcd0 0x420309d9:0x3fcebcf0 0x42030aa6:0x3fcebd20 0x42030b19:0x3fcebd40 0x42030cd1:0x3fcebd80 0x42030d62:0x3fcebde0 0x420309d9:0x3fcebe00 0x42030aa6:0x3fcebe30 0x42030b19:0x3fcebe50 0x42030b8e:0x3fcebe90 0x42030d46:0x3fcebef0 0x4202ff2d:0x3fcebf10 0x4202ffe2:0x3fcebf50 0x42030001:0x3fcebf70 0x4200ad5e:0x3fcebf90 0x420205d2:0x3fcebfb0 0x42037779:0x3fcebfd0

ELF file SHA256: bd2a1aecfa708864
```

redmushie, 2023-10-09:
Try using the ESP-IDF stacktrace decoder. Apparently it's super helpful

redmushie, 2023-10-09:
Also, title only specifies the chip, not the board. @hhvrc

hhvrc, 2023-10-09:
@redmushie turns out its every board :>

redmushie, 2023-10-09:
XD

hhvrc, 2023-10-09:
This memory access violation (Prohibited store operation) occurs here:
https://github.com/OpenShock/Firmware/blob/develop/src/CaptivePortal.cpp#L108-L115

commit fdda6f19 (2023-10-09): Fix sefault on network auth

fixes #40 by implementing a custom string parser
closed 2023-10-09 by hhvrc