OpenShock/Firmware #46: Several "file does not exist" errors upon serving captive portal

redmushie, 2023-10-18:
```
[ 12255][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:10.10.10.11
[ 12256][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 23051][E][vfs_api.cpp:105] open(): /littlefs/www/fa/fa-all.css does not exist, no permits for creation
[ 23098][E][vfs_api.cpp:105] open(): /littlefs/www/_app/immutable/assets/1.css does not exist, no permits for 
creation
[ 23191][E][vfs_api.cpp:105] open(): /littlefs/www/_app/immutable/assets/0.css does not exist, no permits for 
creation
[ 23282][E][vfs_api.cpp:105] open(): /littlefs/www/_app/immutable/entry/4.js does not exist, no permits for creation
[ 23368][E][vfs_api.cpp:105] open(): /littlefs/www/_app/immutable/chunks/9.js does not exist, no permits for creation
[ 23657][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 23658][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 23658][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 23660][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 24080][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 24083][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 24084][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 24084][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 24136][D][CaptivePortalInstance.cpp:37] handleWebSocketClientConnected(): [CaptivePortalInstance] WebSocket 
client #0 connected from 10.10.10.11
```

redmushie, 2023-10-18:
I checked what the `pcb is NULL` error is about but apparently it's safe to ignore.

redmushie, 2023-10-18:
Changed to release 1.2 due to low prio (captive portal works fine)

hhvrc, 2023-10-18:
For context:

This happens because the static file router always looks for the path first, then the path + ".gz"

We only have .gz files, so it will never find the non-gz files.

Its safe to ignore this, but in the future we can easily implement a custom router that tried .gz first, followed by non-gz

PR #89 (2023-11-04): Fix HTTP server crashing due to Watchdog timeout
This change (mostly) fixes the http server crashing due to watchdog timeout (most frequently encountered on mobile).
It will still crash at seemingly random requests, but this should be significantly decreased.

fixes #46 
fixes #80
closed 2023-11-04 by hhvrc