OpenShock/Firmware #47: No UI feedback after connecting to network

redmushie, 2023-10-18:
I connected to the network successfully as I confirmed this via the serial monitor, but there is zero visual feedback in the captive portal that the connection was established.

redmushie, 2023-10-18:
Marked as 1.1.x release, but if opinions differ on this I'm all ears.

nullstalgia, 2023-10-19:
+1 to this, but should make a small step towards it before 1.0 takes flight.

Even if we don't do a whole Succeed/Fail reaction right now, we could do *something* simple like show the newly acquired local IP on the top/bottom of the page if it was able to get one.

More complex animations and try-catches can be done in 1.1+.

hhvrc, 2023-11-01:
This already works now, closing this

closed 2023-11-01 by hhvrc