OpenShock/Firmware #48: One-second vibration lasts nearly 7 seconds

redmushie, 2023-10-18:
```
  Version:  0.0.0-develop
    Build:  debug
   Commit:  70803fb5a1df141dc72d46cd868c40c5c0b4cfc3
    Board:  Pishock-2023
     Chip:  esp32
```

Title.

redmushie, 2023-10-18:
I tested with a shock at 25% and it lasted about 4.3 seconds. Very confusing.

redmushie, 2023-10-18:
It seems to be extremely inconsistent.

[image]

redmushie, 2023-10-18:
This 0.3s shock lasted **9 seconds**: [https://s.redmushie.me/2023/10/VID_20231018_132005.mp4](https://s.redmushie.me/2023/10/VID_20231018_132005.mp4)

hhvrc, 2023-10-18:
This is caused due to a change in the flatbuffer schema corrupting the data read, needs investigating why the flatbuffer validator did not catch this.

closed 2023-10-18 by redmushie

redmushie, 2023-10-18:
@hhvrc I disagree since that messes with future protocol changes. It'd put the padding in the middle.