OpenShock/Firmware #58: [Feature] Show example/pre-set RMT pin definitions on Setup page

nullstalgia, 2023-10-19:
### Pre-submission checklist

- [X] I checked that there is no other Feature Request describing my wish.
- [X] I checked that this feature is, in fact, not supported.
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### What should be possible that currently is not?

If using a PCB/pre-existing design, there should be a placeholder pin set in the RMT pin input on the Setup page.

If not, it should fall back to a known good pin for the board.

### Other remarks

Maybe in Wiki, we can have a listing of recommended RMT pins? Mark out which pins are recommended? (Not used by SPI flash, aren't input-only, etc)

redmushie, 2023-10-20:
Self-documenting UI is better than non-self-documenting UI + separate documentation. And I don't think most of us are excited to document something as basic as this.

Marking 1.0

redmushie, 2023-10-31:
@hhvrc I flashed 1.0.0-rc.2 and found no preset TX pin on the Pishock-2023, which has has pin 12 preset as TX pin.

hhvrc, 2023-11-01:
Actually fixed in #82

PR #76 (2023-10-30): Better device status reporting at local WS connect
fixes #58 
Reports connected pin when captive portal is launched
closed 2023-11-01 by hhvrc