OpenShock/Firmware #59: [Feature] VisualStateManager should be able to handle RGB+Normal LED

nullstalgia, 2023-10-19:
### Pre-submission checklist

- [X] I checked that there is no other Feature Request describing my wish.
- [X] I checked that this feature is, in fact, not supported.
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### What should be possible that currently is not?

On my PCB, it is possible to show a more complex current state of the software.

Possibly using normal LED like other boards do, except commands sent to shockers are displayed on the RGB LED?

### Other remarks

I'll likely be tackling a big part of this, as well as merging in the RGB library I have soon.

redmushie, 2023-10-19:
I'm a little confused about your intent on this issue. It seems a bit.. vague? In part you're alluding to using a GPIO LED + WS2812B LED at the same time but I feel like supporting all three combinations (only GPIO, only WS2812B, both) is _much_ more trouble than it's worth.

I'd personally scope this feature at using one or the other, not both.

redmushie, 2023-10-20:
@nullstalgia Though we won't stop you if you want to implement it yourself 😜.

redmushie, 2023-10-20:
Marking as 1.1

closed 2023-12-01 by hhvrc