OpenShock/Firmware #60: [Bug] Serial commands not working / not saving data / not triggering a connect

LucHeart, 2023-10-20:
### Pre-submission checklist

- [X] I checked that no other Bug Report describing my problem exists.
- [X] I am running the latest stable or prerelease firmware.
- [X] My board is on the [list of maintained boards](https://wiki.openshock.org/hardware/boards/#fully-maintained).
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### Board

Not board specific

### Firmware version

1.0.0-rc.1

### Flashing method

ESP Tool JS

### Describe what happened as precisely as possible.

Serial commands (networks, authtoken) dont seem to work at all.
I cant figure out how to save networks, even after looking into the code I cannot get it to work. The validation also seems to not work since i can just give it bogus data and it will say its okay.

### Describe what you expected to happen instead.

For the commands to work properly as to be expected.

### In as much detail as possible, describe the exact steps you took to make the problem appear.

1. set a network via the networks command `[{"ssid": "myNetworkName", "password": "myNetworkPassword"}]`
2. restart the device
3. see it not being saved

redmushie, 2023-10-20:
Are you having trouble with _all_ serial commands or only the two mentioned?

LucHeart, 2023-10-23:
In addition, i didnt wanna mess with the wifi logic too much as of now. We should probably just figure out why they are not saving correctly for now.

PR #75 (2023-10-30): Fix network configuration via serial
fixes #60 
Code submitted networks to config with a starting id of 0.
Config gave no warning that a id of 0 is invalid.
closed 2023-10-30 by hhvrc