OpenShock/Firmware #61: [Feature] Serial commands to read out exisiting config values

LucHeart, 2023-10-20:
### Pre-submission checklist

- [X] I checked that there is no other Feature Request describing my wish.
- [X] I checked that this feature is, in fact, not supported.
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### What should be possible that currently is not?

We need to be able to read out the current network configuration and rmt pin. Auth Token is not needed and can be write only to strengthen security a bit.

Reason for this need is the flash tool, for it to auto copy wifi and rmt pin config values when updating the firmware for a better user experience.

redmushie, 2023-10-20:
I think this is a good idea and probably a small effort, but as this is non-critical it's perfectly suited for 1.1.

LucHeart, 2023-10-24:
fixed by https://github.com/OpenShock/Firmware/pull/66

closed 2023-10-24 by LucHeart