OpenShock/Firmware #65: [Feature] Support changing OpenShock server at runtime

redmushie, 2023-10-22:
### Pre-submission checklist

- [X] I checked that there is no other Feature Request describing my wish.
- [X] I checked that this feature is, in fact, not supported.
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### What should be possible that currently is not?

Cofiguring the OpenShock server that the firmware connects to should be modifiable at runtime, not compile time.

- Add serial command to change server
- Add (prefilled) text field with baked-in URL as default value

hhvrc, 2023-10-30:
Why is this necessary?

redmushie, 2023-10-30:
Because self-hosting is a primary feature and currently we're hard baking shocklink.net into all binaries.

hhvrc, 2023-10-30:
Due to this being a change being one that is done very rarely I will not be swapping it at runtime, but will implement it as a set-and-reboot procedure, setting it at runtime introduces alot of complexity

redmushie, 2023-10-30:
At a glance, I don't think that wouldn't make sense if you're still on the captive portal, since no gateway connection or auth token was acquired yet.

After thinking on it for a bit, I'm of the opinion that the minimum viable is "set and apply after reboot". Rebooting the board the second you change it will:

1. Boot you out of the captive portal network (if changing it that way), and;
2. Mess with attempts at automating configuration, as the serial connection will probably be interrupted, or at the very least be spammed with a large quantity of irrelevant logs. (Though I admit that's usually of little significance.)

hhvrc, 2024-04-12:
Implemented in #223

closed 2024-04-12 by hhvrc