OpenShock/Firmware #111: [Bug] Gateway pairing fails despite network connection

redmushie, 2023-11-08:
### Pre-submission checklist

- [X] I checked that no other Bug Report describing my problem exists.
- [X] I am running the latest stable or prerelease firmware.
- [X] My board is on the [list of maintained boards](https://wiki.openshock.org/hardware/boards/#fully-maintained).
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### Board

Pishock 2023

### Firmware version

1.0.0-rc.4

### Flashing method

Instructions per RELEASE.md

### Describe what happened as precisely as possible.

UI (these were taken right after each other):

[image]
[image]

Logs:

```
[106470][V][WiFiManager.cpp:289] Save(): [WiFiManager] Authenticating to network ...
[106568][V][WiFiManager.cpp:126] _connectImpl(): [WiFiManager] Connecting to network ... (...)
[106592][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[107951][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: ..., BSSID: ..., Channel: 6, Auth: WPA2_PSK       
[107953][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[107960][I][WiFiManager.cpp:194] _evWiFiConnected(): [WiFiManager] Connected to network ... (...)
```

### Describe what you expected to happen instead.

Gateway pairing should succeed, since internet connection is established

### In as much detail as possible, describe the exact steps you took to make the problem appear.

- Connect to wifi in captive portal
- Try to pair to gateway
- [Get error]

hhvrc, 2023-11-08:
Please provide more logs, this is currently not sufficient for me to troublehshoot off

hhvrc, 2023-11-08:
@redmushie can you update this post so i can remember what to fix?
the ip stuff

redmushie, 2023-11-14:
I think you fixed it but you were supposed to add logging w.r.t. DHCP working or not working

hhvrc, 2023-12-01:
This is a stale issue, if this re-appears again then we would have to look into it

closed 2023-12-01 by hhvrc