OpenShock/Firmware #141: [Bug] Open WiFi networks cannot be connected to

LucHeart, 2023-12-01:
### Pre-submission checklist

- [X] I checked that no other Bug Report describing my problem exists.
- [X] I am running the latest stable or prerelease firmware.
- [X] My board is on the [list of maintained boards](https://wiki.openshock.org/hardware/boards/#fully-maintained).
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### Board

D1 Mini

### Firmware version

1.0.0

### Flashing method

xxx

### Describe what happened as precisely as possible.

Tried to connect the ESP to a open wifi network.

### Describe what you expected to happen instead.

Open networks should be able to be connected to

### In as much detail as possible, describe the exact steps you took to make the problem appear.

- Have a WiFi Network with no password (open)
- Try to connect to it from the captive portal
- Clicking the button does nothing
- Serial-log reveals error

### Other remarks

```
[642589][E][WiFiNetworkSaveCommand.cpp:26] HandleWiFiNetworkSaveCommand(): [LocalMessageHandlers] WiFi message is missing required properties
```

hhvrc, 2023-12-01:
This is not a bug, it's a feature yet to be implemented

LucHeart, 2023-12-01:
it errors, therefor its a bug

PR #164 (2023-12-15): Handle connecting to open (unsecured) networks
Don't know if we want to adjust how we send data in the Svelte as well, or if this is suitable on it's own.

Closes #141 

(Removed TODO: it should be handled in Frontend instead with a manual Add Network option)
closed 2023-12-15 by hhvrc