OpenShock/Firmware #198: [Bug] [CI] Restore old naming scheme for release firmware bin

LucHeart, 2024-02-08:
With 1.1.0 the naming scheme for the firmware merged bin's has changed. It does NOT include the version anymore. This leads to not knowing what firmware version a certain bin is.

This was previously different, dont exactly know why it was changed.

E.g. currently: `OpenShock_Wemos-D1-Mini-ESP32_stable`

What it should really be is something like: `OpenShock_Wemos-D1-Mini-ESP32_stable_1.1.1`
for beta this would be `OpenShock_Wemos-D1-Mini-ESP32_beta_1.1.1-rc.1`

hhvrc, 2024-02-09:
I would change it to ``OpenShock_Wemos-D1-Mini-ESP32_1.1.1`` rather than ``OpenShock_Wemos-D1-Mini-ESP32_stable_1.1.1``

LucHeart, 2024-02-09:
just for stable? would keep for beta thought

hhvrc, 2024-02-09:
This would apply for all release channels

PR #199 (2024-02-10): Include firmware version in merged CI file
Merged binary will now be named like this: ``OpenShock_Wemos-D1-Mini-ESP32_1.1.1``

closes #198
closed 2024-02-10 by hhvrc