OpenShock/Firmware #234: [Bug] Frontend becomes non responsive when changing tx pin to invalid

LostQuasar, 2024-05-15:
### Pre-submission checklist

- [X] I checked that no other Bug Report describing my problem exists.
- [X] I am running the latest stable or prerelease firmware.
- [X] My board is on the [list of maintained boards](https://wiki.openshock.org/hardware/boards/#fully-maintained).
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### Board

DFRobot Firebeetle

### Firmware version

0.0.0-develop

### Flashing method

Used platform.io to flash firmware

### Describe what happened as precisely as possible.

Web UI became completely non responsive when an invalid pin was selected via the web UI. Refreshes caused no response from the device.

### Describe what you expected to happen instead.

An error to pop up and continue to work

### In as much detail as possible, describe the exact steps you took to make the problem appear.

1. Change Wifi network (maybe not required)
2. Change to invalid pin (below 11)
3. Refresh

hhvrc, 2026-11-03:
fixed since 1.4.0 (in #269) lmao

closed 2026-11-03 by hhvrc

LostQuasar, 2026-11-03:
lmao lol even