OpenShock/Firmware #282: [Feature] Option to use Channel 2 or 3

bemabalu, 2024-09-07:
### Pre-submission checklist

- [X] I checked that there is no other Feature Request describing my wish.
- [X] I checked that this feature is, in fact, not supported.
- [X] I accept that this issue may be closed if any of the above are found to be untrue.

### What should be possible that currently is not?

If using the devices not only with OpenSchock but with a remote as well it would be great to use 1 ID on all 3 Channels.
Reading the ID from the remote can be done with [https://github.com/CrashOverride85/collar](https://github.com/CrashOverride85/collar ) for example.

hhvrc, 2024-09-07:
We have considered this and the protocol encoders support it, we just havent found a use-case for it yet.
I'm not sure i understand you correctly, why do you want to use the channels feature?

bemabalu, 2024-09-07:
Well, i want to use the shock devices with open shock and the remote control simultaneously. 
This works fine on Ch1 alredy. Ijust read out the Remote ID Adress and set this in Openshock.
If there would be an option to use the other channels as well in Openshock you could use one remote with 3 shock devices and trigger them by remote OR openshock without the need to pair them again and again.

One more remark on the tool to read the remote ID i mentioned. I found out it dekodes MSByte and LSByte in the wrong order atm. On my fork of the ropository i fixed this issue.